S-March/smarchWatch_Public #1: STEP/STP or IGES/IGS housing files?

justinclift, 2019-04-30:
Pretty nifty project. :smile:

With the housing files, any chance of STEP or IGES?

STL's are good for some things (eg 3D printing), but STEP or IGES include the model data, so are better for interchange with other cad software.

eg people can productively load a step or iges file in (say) FreeCAD and usefully modify it. :smile:

amishHammer, 2019-04-30:
Even and export of the Fusion F3D file would be helpful!!

vanbluca, 2019-05-01:
You can use the grabcad workbench to export the files in other formats really easily.

S-March, 2019-05-01:
I should be able to export them later tonight and put them up here!

S-March, 2019-05-01:
Didn't mean to close it. I will upload the files later tonight.

amishHammer, 2019-05-10:
> You can use the grabcad workbench to export the files in other formats really easily.

Yeah that's only good if you have a real solid model to put into grabcad.  With just the step file you can't do much conversion.

justinclift, 2019-05-10:
> With just the step file you can't do much conversion.

Typo?  The step file should be a solid model, whereas the stl file isn't real useful.

Well, as far as I understand anyway. :smile:

S-March, 2019-05-10:
Sorry I am just getting to this. The step files are now committed.

closed 2019-05-10 by S-March

justinclift, 2019-05-10:
No worries at all @S-March, thanks for taking the time to get this stuff online. :grinning: