S-March/smarchWatch_Public #10: Boom!

zerosimms, 2019-05-01:
This looks awesome! 
Thank you so much for sharing and congrats on the Engadget publication!

Just one question, for the PCB board, can you tell me roughly how much it will cost? Or where I should begin to find out for myself - I can see it's not an "off the shelf" component and I'm assuming I would need to provide the PCB files to a manufacturer somewhere?

Sorry for the newbie question but normally I buy off the shelf, this part is all new to me.

Cheers

vanbluca, 2019-05-01:
I am looking into ordering a larger amount of PCBs at once to cut the costs. I already got some interested people. I expect 2pcs will be about $10 including shipping. Let me know in case you want one (or a few). I already got some interested people.

zerosimms, 2019-05-02:
Thanks, @vanbluca - I'd be very interested in purchasing, I'll wait for you ebay link :) Cheers!

closed 2019-05-02 by S-March