S-March/smarchWatch_Public #12: Seperate repository for Hardware, Software and other files

anselm94, 2019-05-01:
My suggestion will be to ***create*** an **Organization**

  - Interesting name (like Pebble) for the Organisation [Tried few suggestions here](https://namelix.com/app/?keywords=walnut+watch+smart+time+color+love)

  - Create separate repositories for
    - GitHub pages website
    - Hardware
    - Software 
    - Body Design
    - miscellaneous and tutorials
    - etc.

anselm94, 2019-05-08:
#28 Can we have seperate repository for iOS project and create a starter project for Android, for people to start contributing?

S-March, 2019-05-10:
This is a great idea. I will look into it when I have a chance. I will close the issue for now, but please feel free to open another in the future.

closed 2019-05-10 by S-March