S-March/smarchWatch_Public #15: Wish list

S-March, 2019-05-01:
I thought it might be worth adding a wish list here for features that people might want. I may not have time to implement all of them, but maybe others in the community can!

vanbluca, 2019-05-01:
Would love a video on how to put your codes in the actual PCB.

Piflyer, 2019-05-01:
Hi, first off, congrats on creating this smartwatch! I lost mines, and I am desperate to find a replacement. I do like doing DIY crafts, but I would like to know if you could also support Android notifications. I think this would be very helpful since this project would then be supported by both platforms. I was wondering in the future, do you have any plans to make this more like Wear OS? I think that would be something very interesting to see happening. Thank you and have a great day!

abhinuvpitale, 2019-05-01:
Android Support?
Adding support for a touchscreen display - Resistive possibly?

anselm94, 2019-05-02:
Possibly a single framework for the watch. Then Android & iOS apps to communicate with the watch?

Anders0lesen, 2019-05-02:
This project certainly has my attention, but I would love to be able to use this as a G-Pay platform, which is pretty much the only reason I even am considering having a SmartWatch in the first place...

syphen, 2019-05-02:
Android Wear devices specifically use Snapdragon Wear chipsets. Would there be any advantages to building a device like this with something like the Snapdragon 670? (I don't know much about the hardware limitations or restrictions)

Also...swap out your display for a Transflective LCD so you can have the time visible with the screen "off".

trustbyte, 2019-05-02:
congrats for this project and making it opensource.

the biggest plus is i believe the software. clean and simple, no java, no siri, no bullshit.

if i were you I would try to get in touch with the guys that make the Raspberry. make it happen as a kit(no soldering), donate a part of the money and get some for yourself as you deserve it and let us get some fine smartwatch. 

tutorials at all levels would be great to many. your work is awesome but many maybe can't get their head around the files and code without
some more details and clarifications.

the battery life of one week is great and while something life pulse sensor or anything health realated is nice it should stay away if the battery becomes 2 days max.

keep yourself healthy and don't stop innovate, we are grateful and courious of what can come next.

S-March, 2019-05-02:
Thank you so much!

I will try to provide some more guides and better descriptions of some of
the neats parts of the code in the coming days (if I have some time). I
really appreciate the thought about monetizing this, but I really like the
idea of it being free for anyone out there! As a kid I was always bummed
when I couldn't afford to work on a cool project.

Stay healthy yourself and I can't wait to see what awesome stuff YOU come
up with next! We are all creators and innovators in our own way.

Best,
Sam

jurkov, 2019-05-02:
Thank you for your work!
Is there enough space to add a LoRa IC for LoRaWAN communication?

adrienlesage2005, 2019-05-03:
Touchscreen or physical buttons, custom watch face support, music control, calculator, and a nice UI. Basically an open source Pebble killer :P

oeah2, 2019-05-05:
@S-March 
I am an (junior) electrical engineer too and I like your idea.
I'd like to contribute to the project. Can we get in contact for this?

nh1402, 2019-05-05:
support for a bigger battery would be nice, the BL-S6 is the largest smartwatch battery that I'm aware of at 570mah (the LG Urbane 2nd Edition LTE battery), although it runs at a higher voltage at 3.85v.

pratikshiyal, 2019-05-06:
Would it be possible for you to share not just the gerber files but also the PCB and Schematic design files? this would enable the community to make any hardware modifications they like.

alextechcc, 2019-05-08:
Can you direct me to any materials you referenced / know of for that crazy antenna design?a

absint0o, 2019-05-16:
Heart rate monitor please!!! Congrats on the watch! A true open source flagship killer!

nicholaswagner, 2019-08-04:
A heart rate monitor would be awesome, but this project is fantastic!  Thanks for sharing!

dannygrob, 2019-08-30:
I'd love to have an extra connector for CTP on this display:
https://www.ebay.com/itm/1-3-inch-240x240-round-IPS-SPI-tft-lcd-display-with-CTP-for-smart-watch/112800854794

This wil open up so much more potential

boelle, 2020-07-27:
> This project certainly has my attention, but I would love to be able to use this as a G-Pay platform, which is pretty much the only reason I even am considering having a SmartWatch in the first place...

same thought here, why a watch when the phone can tell the time :-D

boelle, 2020-07-27:
> Touchscreen 

would be usefull

boelle, 2020-07-27:
space for a rubber ring/seal between display and case and another one between the bottom and the case

would not make it a dive watch but would keep it more safe

boelle, 2020-07-27:
working with android would also be nice

> Android Wear devices specifically use Snapdragon Wear chipsets. Would there be any advantages to building a device like this with something like the Snapdragon 670? (I don't know much about the [...]

neonot, 2020-11-13:
Now that you have got a smartwatch and smart-glasses. How about going smaller and make a smart attachment to go between conventional watch lugs and a strap? I have been toying with this idea for quite some time but not getting anywhere and looking for a buddy to discuss. This project would need a small OLED screen with intelligence and Li-poly battery. Something like [this but not fixed ](https://youtu.be/Y7s2CCg0KZY?t=508)to a band. I think it needs to go between the band and watch lugs (near side).  The hardware should all be available. The battery and charging circuit could slide under the actual leather or steel  strap/bracelet of the watch on the near side. One could enjoy both conventional watches with a smart OLED notification exactly where it is best visible.

Nightchicken, 2023-07-15:
I absolutely love this project and find it absurd that you spent so much time on this.

An app ecosystem, maybe like a git package thing to do it. 

Also a dial like what the apple watch has for menu navigation.

I am definitely programming tetris for it when I get the chance.

Thank you for all the time you have spent on this project and then making it open source.

Whiteeeey, 2025-05-18:
I think this is a fantastic project that demonstrates what can be done with some simple equipment. Unfortunately, the ANCS notifications no longer work with ESP32 3.02 libraries, something to do with different advertisement and even when adjusted to allow the code to compile the esp32 does not appear on apple phones as a device but it does appear on android without the possibility to connect. I'm trying to see if I can get it working with the old bluetooth libraries but with little success. Has anyone else run into these issues or is there a way to easily work around this?