S-March/smarchWatch_Public #8: Not enough info on PCB

vanbluca, 2019-05-01:
When I try to order the PCB I need to fill in a lot of info I can’t find anywhere (like thickness). Could you give us some more info on how to order the PCBs?

S-March, 2019-05-01:
Absolutely. Let me see if I can just take a screenshot of the parameters and upload it.

S-March, 2019-05-01:
There should be a picture with board parameters in the PCB folder now.

S-March, 2019-05-01:
I'm going to re-open the issue so people can see, I added a link to the main page where you can just have the same pcb maker make your boards for you. I set up all the files and parameters already, so you can just click "order".

https://www.pcbway.com/project/shareproject/smarchWatch.html

vanbluca, 2019-05-01:
I will order 100pcs and sell pcb kits (pcb for watch + pcb for charger) for $12 on eBay. I was thinking $2 of every sale will be donated to the maker. Worldwide shipping is $1.50, so total price is $13.50

Feel free to announce this on your github info page too.

S-March, 2019-05-01:
You're too kind! I will pin this issue and close it out so others can see it.

closed 2019-05-01 by S-March

vanbluca, 2019-05-01:
This project is so great! I want to help as many people as possible recreating this!

vanbluca, 2019-05-01:
Perhaps you could announce it somewhere under “code” too? I can imagine not everyone reads “issues”

vanbluca, 2019-05-01:
I meant in the README.md document

abhinuvpitale, 2019-05-01:
Is this legit?

vanbluca, 2019-05-01:
Update: will be selling through eBay so you guys have buyer protection!

anselm94, 2019-05-02:
@vanbluca Please post the link once the eBay listing is ready. Fans like me are awaiting here 😃

edwinchenloo, 2019-05-02:
shouldn't you give time for a discussion of version 2 before ordering pcb's?   I for one would like an optical heart rate sensor.  Also, of more value to people would be the selling of already populated pcb's.  I don't have the soldering skills and equipment needed to do that.

vanbluca, 2019-05-02:
@anselm94 I will order them later this week an expect to have it in about 2 weeks from now. Will keep you posted

vanbluca, 2019-05-02:
@edwinchenloo I personally don’t think the maker has time to make a version 2 anytime soon. I may be offering fully soldered PCBs later too, but these will be much more expensive (an additional $25 dollars for the components i have to solder on it and I will have to charge you for the time I spend on soldering)

vanbluca, 2019-05-02:
PCBs have been ordered

abhinuvpitale, 2019-05-03:
I agree with @edwinchenloo, I think there is a need for soldered components rather than bare PCBs, because that's thekey bottleneck here.

exploshot, 2019-05-03:
Will you add a Link here? Total costs then 40$?

vanbluca, 2019-05-03:
@abhinuvpitale I guess I will be offering soldered PCBs too, but that will take some more time since I am currently quit busy

vanbluca, 2019-05-03:
@exploshot I will definitely add a link here to the listing for the $12 PCBs. As for the $40 did you mean that is for the unsoldered PCBs + spare components or for the fully soldered PCBs? I hope you understand I can’t solder it for $3...
I am not sure yet if I will be offering an option to purchase bare PCBs + the required spare components. This is because I would have to order lots of components with the risk of ending up not selling them due to lack of interest.
You can also simply order the components here: https://www.digikey.com/short/pzz59b
(Please note: battery not included!)

Blue-Dust, 2019-05-05:
I am interested in building one, but how experienced do you need to be for the soldering?  I am expecting it to be quite small/intricate job.  I am not ready for that type of work yet.  Also, I read somewhere this is only working on iPhones (post/comment).  I have an Android phone and would hold off until I knew it was compatible.  I would also want to buy a kit, or easy links to order everything so I don't buy the wrong stuff.  Good idea on selling the boards on eBay.  I would be more willing to buy into this project with that option. Is there a video showing how to use it and it's current features?

vanbluca, 2019-05-06:
@Blue-Dust as for the soldering, it is not like soldering with a soldering iron. This video shows the technique used for it: https://youtu.be/qyDRHI4YeMI. The watch is indeed currently for iPhones (I assume since the maker @S-March has an iPhone and not and android phone). Perhaps @S-March decides to make and wondrous version too (if he has time) since lots of people mentioned (in issue #15) they would really like a version that supports android.
As for kits/links please wait a little longer and I will prepare this for you guys. PCBs will listed on eBay once I have them (I expect in about 2 weeks).
All the pics, vids and info that are out there are in this github page and the imgur page. As for the current features it is pretty straightforward; it is basically a digital watch that forwards notifications from your phone.

vanbluca, 2019-05-06:
Here are some links for the components:

1. Electronics for PCBs (watch+charger): https://www.digikey.com/short/pzb540

2A. Battery (US only)(recommended): https://www.digikey.com/products/en?keywords=1572-1630-ND
2B. Battery (outside US & cheaper alternative): https://m.dhgate.com/product/5pcs-1-lot-lir2430-3-6v-lithium-li-ion-rechargeable/409686938.html?invitorid=3uqYbq50

3A. Display (eBay): https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F112839389316
3B. Display (DHgate): https://m.dhgate.com/product/aml130a2402-1-3-240-240-inch-round-tft-lcd/413657684.html?invitorid=3uqYbq50

amerello, 2019-05-14:
@vanbluca I don't think the hot air gun method would work with the AQFN60 package of the DA14683 SoC. I wonder if the [skillet method](https://hackaday.com/2013/07/28/electric-skillet-reflow-soldering-guide/) would be a better option if you don't have a proper reflow oven, but you don't really get to see when the pins have been properly soldered, so it seems to me that luck would in any case play an important role.

vanbluca, 2019-05-14:
@amerello did you realize you have to solder 2 sides of the PCB? That made me think the skillet method is less of a better option.

amerello, 2019-05-14:
But the backside is quite simple, i think it could be done on a second stage with a soldering iron, right?

vanbluca, 2019-05-14:
A soldering iron? Probably not. The second stage contains among other things a vibration motor. This vibration motor has 2 contacts that are each 0.5mm wide. I don’t think a soldering iron will do the job...
I would recommend buying a $200 reflow oven from China or a $30 “rework” hot air gun if you have a smaller budget. I am confident the hot air gun could do the job as well as a reflow oven as long as you set the air temperature and airflow right and as long as you keep moving the hot air gun. Never keep the hot air gun at the same place!

S-March, 2019-05-14:
That is correct @amerello 

In fact that is how I put together my few boards. I used a reflow oven for the top side, and hand soldered the bottom. The hot air or skillet approach should work on the top side. If you do hot air, just make sure you put some solder down on the pads before you try and put the microcontroller down.

S-March, 2019-05-14:
@vanbluca the pads for the vibration motor, although small, are pretty forgiving. There are only two pads so you can put the part down, if it is not centered nicely, just touch your iron to the pads and they should be ok.

vanbluca, 2019-05-15:
Good news! I just received the PCBs and listed them on eBay. Here is the link: https://ebay.us/FQBGIs
Unfortunately, eBay lets me only list/sell a total of 50 items each month, so I wasn't able to list my entire "stock" yet...

vanbluca, 2019-05-16:
@S-March could I get something like your PayPal email for the donations?

S-March, 2019-05-16:
@vanbluca You're too kind. Can you DM me on reddit and I can send you the email address?

vanbluca, 2019-05-17:
@S-March done!

I have to say I was hoping I had sold a little more by now, but perhaps I will have this weekend.

trisimix, 2019-05-17:
I'm gonna buy one @vanbluca some people I know too maybe

trisimix, 2019-05-17:
Just need to figure out everything else I need, the electronics obviously, but the tools needed to place them as well

pratikshiyal, 2019-05-18:
@vanbluca  the listing in eBay has been removed? Or are you out of stock?

vanbluca, 2019-05-18:
@pratikshiyal no I am not out of stock and the listing is still online. Perhaps this link works: https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F254232630219

vanbluca, 2019-05-18:
In case you can still not find it:
1. Go to eBay.com 
2. Search: smarch watch 
3. Check eBay doesn’t auto correct it to “smarth watch” or so
4. The listing should appear now

vanbluca, 2019-05-18:
@trisimix that is great to hear! As for the components and tools, here are some suggestions. When there are more options (for example A and B), you can choose either one of them.

Here are the components I ordered:

1 Electronics for PCBs (watch+charger): https://www.digikey.com/short/pzb540

2A. Battery (US only)(recommended): https://www.digikey.com/products/en?keywords=1572-1630-ND
2B. Battery (outside US & cheaper alternative): https://m.dhgate.com/product/5pcs-1-lot-lir2430-3-6v-lithium-li-ion-rechargeable/409686938.html?invitorid=3uqYbq50

3A. Display (eBay): https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F112839389316
3B. Display (DHgate): https://m.dhgate.com/product/aml130a2402-1-3-240-240-inch-round-tft-lcd/413657684.html?invitorid=3uqYbq50

As for the display, I recommend to go to the eBay listing and ask the seller to ship it as a letter in an envelope since this will cut shipping costs, import charges and administration costs for the import charges.

Tools:
 
1A. Reflow oven (for soldering):  https://s.click.aliexpress.com/e/OP5p41xpv
1B. Rework hot air gun (cheaper alternative for reflow oven): https://s.click.aliexpress.com/e/bNyNkmh5p
1C. Skillet method (see message from @amerello above): 

2 Solder paste: https://s.click.aliexpress.com/e/kgqmfXvGX

vanbluca, 2019-05-18:
I forgot to mention, but I will be abroad between May 23 and May 30. I won’t be able to ship anything between those 2 dates.

kamshee, 2019-06-11:
@vanbluca Any option for soldered PCBs? If not are the stencils available with the pcb listing? Thanks.

abhinuvpitale, 2019-06-19:
> @trisimix that is great to hear! As for the components and tools, here are some suggestions. When there are more options (for example A and B), you can choose either one of them.
>
> Here are the [...]

The digikey list is not showing the right components, it shows incorrect part number, could you please check that?

Ajayrulez, 2019-07-20:
This is great. I got all of BOM ordered and shipped to LA.

saurabhreigns, 2019-08-29:
Electronics for PCBs (watch+charger): https://www.digikey.com/short/pzb540
Checked the list items and I am unable to find U1 (in bom designator)  it is supposed to be a 24 pin receptacle for LCD which is to be soldered onto the PCB. I ordered these items form digikey: 
1. CONN RCPT 24 POS SMD Gold H122178DKR-ND 
2. CONN RECEPT 10 POS 0.4mm SMD -BM14B(0.8)-10DS-0.4V(53)CT-ND 
and the parts i received are not correct. they have 28 and 14 pins without metal fittings. Moreover , under a microscope i could see that they will not mate with the lcd connector. Can someone please help with a link to purchase correct female mating connector for the lcd. Supplier from India will be preferred. 

[Images.zip](https://github.com/S-March/smarchWatch_PUBLIC/files/3554406/Images.zip)

dannygrob, 2019-08-31:
I think the display connector is: ‎H11850CT-ND‎ from Digikey

d3mon187, 2019-12-11:
> 
> 
> Good news! I just received the PCBs and listed them on eBay. Here is the link: https://ebay.us/FQBGIs
> Unfortunately, eBay lets me only list/sell a total of 50 items each month, so I wasn't able to list my entire "stock" yet...

@vanbluca  Haven't heard from you on ebay, and was wondering if you missed my pcb purchase, or managed to ship it out yet?  I'm not an avid ebayer so I would understand.  Thanks!

darshanys, 2020-02-13:
Is anybody selling populated PCB board yet?

vm41, 2020-10-22:
@S-March can you mention values of pi network components?

S-March, 2020-10-22:
@vm41 no need for any. You can drop in a 0ohm resistor in the series element or just blob some solder and make a bridge!

vm41, 2020-10-22:
> @vm41 no need for any. You can drop in a 0ohm resistor in the series element or just blob some solder and make a bridge!

Ok. Thank you!

vm41, 2020-10-26:
@S-March  Are the board parameters for charger and watch the same?

S-March, 2020-10-26:
@vm41 It's a 2 layer board. But otherwise, yes.

vm41, 2020-10-29:
> I'm going to re-open the issue so people can see, I added a link to the main page where you can just have the same pcb maker make your boards for you. I set up all the files and parameters already, so you can just click "order".
> 
> https://www.pcbway.com/project/shareproject/smarchWatch.html

Are charger files also included in above link?

S-March, 2020-10-29:
@vm41 No. Unfortunately, I could only set 1 board per project.