S-March/smarchWatch_Public #16: Freediver and scubadiving watches

juliankauai, 2019-05-02:
Watches for underwater use seem way overpriced and lack normal smart watch features. Regular companies like Apple and Google aren't likely to near guaranting the depths that Freedivers dive to generally 100 to 400 ft. We have always wanted more sound alarms, brighter displays and features especially for freedivers and scuba divers.

Please consider work/experimentation in this area. Currently 99% of watches are old style and lack any smart watch features. 

Thanks, it's wonderful to hear about your project.

edwinchenloo, 2019-05-02:
For scuba purposes, one better rely on something commercial.  Why risk relying on something homemade while at great depths?  Having said that, finding a 25-50m waterproof case would make it ok for recreational swimming and surfing.  The original case used wouldn't even be ideal for when showering or light rain.

boelle, 2020-07-27:
my comment here might solve the issue with light rain and showering: https://github.com/S-March/smarchWatch_PUBLIC/issues/15#issuecomment-664059135

of course case has to be wathertight itself