S-March/smarchWatch_Public #17: is there an method to emulate ?

MrGate, 2019-05-02:
Is there an way to emulate it using QEMU or something custom to speed development cycles? 

I was just thinking, this would help developers who want to come to the project but don't have the resources to build a physical piece of hardware yet!

S-March, 2019-05-02:
You can buy a development kit from dialog that has the microcontroller and built in programmer. You would have to change the memory, but if people are interested, I can provide a how-to on that. It is what I used before getting my custom hardware in.

abhinuvpitale, 2019-05-03:
Definitely Interested in this!

MrGate, 2019-05-05:
i'll be following this, it will be about a month until I would have extra funds to do this if I wanted to contribute to this project, could you build me one of your units and I could pay for parts + shipping?