S-March/smarchWatch_Public #18: Tips for finishing of 3D printed wood parts

vanbluca, 2019-05-02:
Some of you may be interested in printing the components in wood too. I personally like the finishing on this 3D print: https://imgur.com/gallery/yJAAS

Thought I should share it with you guys since it may be useful.

Feel free to post more tips for the finishing below

vanbluca, 2019-05-02:
It may also be cool to make the watch in metal and apply a cool finish. You would possibly have to apply some sort of isolation to the inside of the watch to avoid shortcuts, but that shouldn’t be too hard.
I.materialise offers an option to order components in all kinds of metal (even in 18K gold!). I personally like the brass option since they are offering really good lookin surface finishes (18k yellow or red gold plated, chrome plated, rhodium plated and (perhaps one of my favorites) black color plated. You can read more about it and find examples right here: https://i.materialise.com/en/3d-printing-materials/brass/colors-finishes

Unfortunately, ordering them in metal isn’t exactly cheap. For instance, the watch body top would be €99.01 and the watch body bottom would be €67.94 excluding VAT. when ordering them in brass (with black color plating and polish). Since the company is based in Belgium, the VAT should be 21% (I assume they apply this on orders from all over the world). This would make the body of the watch cost €166.95*1.21= €202.01 (about $225).
This is indeed expensive, but I believe it brings the watch to a whole new level!

S-March, 2019-05-02:
You also would need to be careful with the antenna. You may end up attenuating too much signal with a metal case.

skintigh, 2019-05-02:
Did you consider making them out of actual wood on a CNC machine?

S-March, 2019-05-02:
If I had a CNC, I wood** have!

skintigh, 2019-05-03:
I had access to one until mid April... Hopefully in September I will again and could give it a shot.

Will you be selling kits?

vanbluca, 2019-05-03:
@skintigh I will be selling kits for the bare PCBs (I ordered 100pcs so you can all benefit from bulk prices). As mentioned before, $2 of every sold lot will go to @S-March to support him. You can read more about it in issue #8. I will maybe also offer soldered PCBs at a later stage for an additional charge since lots of people seems to be interested in this.

With the fully soldered PCBs you will only have to order the screen, battery and make a casing (I could 3D print it too on request). Then all that is left to doo is put the coding in the PCB and assemble it. I think fully soldered PCBs are already quit close to kits...

davidkraus13, 2019-05-05:
Hi, I want to try make one housing at CNC, can you share 3D model in something else then .stl? it will be better for CAM preparation.

vanbluca, 2019-05-05:
@davidkraus13 please download the STL files. Then upload the files to your grabcad workbench. From there you can export it in the format you want.

davidkraus13, 2019-05-05:
Unfortunate from STL is conversions bit of pain, but I will find way

vanbluca, 2019-05-05:
@davidkraus13 what file format do you want?

davidkraus13, 2019-05-05:
STEP will be perfect, or original file from your cad, Im using Rhinoceros so Im able open lot of files. Unfortunately STL is similar like you save DWG drawing to JPEG you make "raster" from model. It works fine for 3D print, but for CAM purposes is far from optimal.

vanbluca, 2019-05-05:
I will try to convert it for you later today. 
For the original file from CAD only the maker @S-March will be able to help you

davidkraus13, 2019-05-05:
Hi, Finally I decide to remodel housing, I pull request to merge to master. Inside features shut be same, bit different outside design.

S-March, 2019-05-10:
Original cad step files are now committed.

closed 2019-05-10 by S-March