S-March/smarchWatch_Public #19: Battery only ships in US (solution)

vanbluca, 2019-05-02:
The battery RJD2430C1 is manufactured in the US and can’t be shipped outside the US due to restrictions. This battery should work as well (but unfortunately they have a smaller capacity): https://m.dhgate.com/product/5pcs-1-lot-lir2430-3-6v-lithium-li-ion-rechargeable/409686938.html?invitorid=3uqYbq50
Any other “LIR2430” battery that is 3.6 or 3.7V should work too

S-March, 2019-05-10:
I am not entirely sure this works. The part that I linked has two "legs" that are used to solder to the underside of the board. I would order the Illinois capacitor part to be sure.

closed 2019-05-10 by S-March

vanbluca, 2019-05-11:
I would defenitly prefer to order the Illinois capacitor battery as well, but that is not an option for everyone. The battery contains lithium which leads to shipping restrictions and therefore they won't ship it outside the US. For those we live outside the US, we need to find a solution and ordering a battery from china/hong kong seems to be the best options
I guess we would have to make the "legs" ourselves. This could be made out of a very thin tin sheet. You may need to isolate some parts of the legs or battery in order to avoid shortcuts. This can easily be done with nail polish.
Another option is that someone located in the US orders a bunch of these batteries and ship them as a letter (probably without tracking code) outside the US. I would highly appreciate it if someone could order one or a few pcs for me and ship it in an envelope to The Netherlands. For contact: l.vanbeers@icloud.com

jspoe, 2019-08-01:
+1 Would like to know where to get this battery in europe too, of have someone to ship it to me from the US.

jspoe, 2020-01-27:
For those who are looking for someone who ships the listed battery outside of the US (ground shipment): 

https://nl.aliexpress.com/item/4000548851959.html?spm=a2g0o.productlist.0.0.3b4535467OV3DN&algo_pvid=8ce8d697-7ef4-466f-a810-93ae19ab986d&algo_expid=8ce8d697-7ef4-466f-a810-93ae19ab986d-0&btsid=30d8a4b4-d0c9-4ed2-9cf6-648eacb5ee1c&ws_ab_test=searchweb0_0,searchweb201602_3,searchweb201603_53