S-March/smarchWatch_Public #22: Make a video showing the watch

8lall0, 2019-05-03:
I would love to see a video with the smarchwatch in action!

vanbluca, 2019-05-03:
I recommend to mention this in issue #15 (wish list) since that topic is meant for it.

TomasHubelbauer, 2019-05-03:
Seems to me like that issue is meant for features, this is more of a meta-issues relating to the repository IMO.

vanbluca, 2019-05-03:
You could be right. I personally thought it was for both

S-March, 2019-05-03:
Hey! Sure thing, do you want something different than what is in the gifs in the Imgur album?

Not much more functionality to show other than notifications and time...

closed 2019-05-10 by S-March

8lall0, 2019-05-10:
Hi! sorry for being late :)

I didn't see the GIFs, their play was blocked by my phone.
They are enough, so good call on closing this "issue".

Thank you and apologies.