S-March/smarchWatch_Public #23: Not enough info on components for charger

vanbluca, 2019-05-03:
Some of the components for the charger are not listed in the “bill of material”. For example the magnet (locks watch in place) and the (female) connector that is for the cable that goes into the charger. Perhaps other components for the charger PCB are also missing, but I am not sure about that.
Could you please add the missing info? As for the magnet could you give us at least the dimensions?

vanbluca, 2019-05-04:
According to the “schematics” files, you will need the following additional components for the charger (all just 1 piece):
10118194-0001LF (609-4618-1-ND) 
ESD9B5.0ST5G (ESD9B5.0ST5GOSCT-ND)
825-22-002-10-001101 (ED10660-ND)

I recommend to wait with ordering them until @S-March confirmed. Perhaps @S-March could also tell us the dimensions of the magnet he used.

S-March, 2019-05-10:
Those are the correct parts. Also this is the magnet I used:
https://www.amazon.com/gp/product/B074BP1FR8/

closed 2019-05-10 by S-March

vanbluca, 2019-05-11:
I was surprised to see you used a 32mm magnet while the space in the body is just under 30mm. I would personally go with a 30*5mm or a 25*5mm magnet.

S-March, 2019-05-11:
There is no magnet in the actual watch. It is just in the charger.

vanbluca, 2019-05-11:
Yes I know. I was talking about the space in the housing of the charger...