S-March/smarchWatch_Public #24: BEWARE: POSSIBLY WRONG SCREEN LISTED IN BILL OF MATERIALS

vanbluca, 2019-05-03:
According to the “bill of materials” you will need a AML130A2402A screen. If you search for this model, you will probably find this one: https://m.dhgate.com/product/aml130a2402a-1-3-inch-240-240-round-tft-lcd/413771628.html?invitorid=3uqYbq50
In one of the gifs/videos from the Imgur page, you can see the used LCD screen (see attached picture). When you have a closer look at it you see it has only one connector instead of 2. This would mean the screen listed as “AML130A2402A” is the wrong model since that one has 2 connectors (and some other small differences). I think the right screen is being sold as “AML130A2402” (not the “A” at the end). Here is a link: https://m.dhgate.com/product/aml130a2402-1-3-240-240-inch-round-tft-lcd/413657684.html?invitorid=3uqYbq50

I recommend you guys not to order any screens until @S-March has replied to this. In case you already ordered the screen you may want to consider canceling it (if possible).
[image]

S-March, 2019-05-03:
Hmm... 

The display manufacture (Amelin) lists it with the “A” at the end in their catalog. 

http://amelin.com.cn/en/pd.jsp?pid=381&mid=339

But you are correct, you want the one without the touchscreen/extra connector.

vanbluca, 2019-05-03:
Yeah that’s weird. 
Anyways, these should be the right ones:
https://m.dhgate.com/product/aml130a2402-1-3-240-240-inch-round-tft-lcd/413657684.html?invitorid=3uqYbq50
https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.com%2Fulk%2Fitm%2F112839389316

vanbluca, 2019-05-03:
@S-March it would be nice if the watch supported touchscreen though.

S-March, 2019-05-03:
I made a conscious choice not to include it. I felt that touch just doesn’t work for a screen so small. If others want to include it, that it totally up to them, that’s why the files are open source! :)

vanbluca, 2019-05-03:
I have to say I agree. Btw, could you please also have a look at issue #23

closed 2019-05-10 by S-March

vanbluca, 2019-05-16:
Tip for ordering the LCD screen:
I recommend to contact the seller and ask to ship it in a letter. This cuts both the very high shipping costs ($41) and import charges. Also their shipper (DHL) asks $20 administration costs for the import charges. This means you would have to pay $10+$41+$20+import charges for only 1 LCD. When they ship it in a letter, shipping will definitely be way less than $41. Also most companies put a value like $5 on an envelope so you won’t have to pay any import charges and administration costs!