S-March/smarchWatch_Public #25: Loading sofware

amerello, 2019-05-03:
Congratulation @S-March on this awesome project. I've looked at the very inspiring blog that you wrote and the repository, but I wasn't able to see how you transfer the software to the watch. Through the charger interface you only provide VBUS. Is this done via BT? If so how do you activate/deactivate the programming mode?
Many thanks again for the awesome write up.

vanbluca, 2019-05-03:
@amerello I totally agree! In issue #15 I mentioned I would highly appreciate it if @S-March could make a video on how to put the coding “in the PCB”.

bdcooke, 2019-05-16:
I too, would love to know how to program this. We got the code. great, I have no idea how send it to the board.

abisheksethu, 2019-05-18:
As far as I understood, the watch schematic explains about header pads which have UART TX and UART RX (connected to B23 and A35 of DA14683). Using that, we can flash binary (QSPI built) to the memory device using the dialog software tool via USB. Also, we have to use the respective voltage levels for connections (connectors/cable), like A35 and B23 might be 3.3V (TTL range) and USB of PC might be different.

amerello, 2019-06-12:
Thanks @abisheksethu that makes sense! That also means that the programming interface USB to VBUS, RX, TX and GND is not yet documented in this repository. RX, TX and GND are on the top left corner of the first layer, that would probably be the place for a connector

jacobandersson86, 2019-08-15:
I cannot manage to upload my binaries. I'm trying to do this with serial. I'm using at FTDI and the TX RX pads on the PCB.

I'm trying to flash using the "External tool" called "program_qspi_serial_win".  The console tells me it can upload the bootloader but it cannot verify RAM.

If I set up a serial terminal (in my case PUTTY) I get the message "DA14683" written 4 times, just as described in the getting started tutorial on Dialoge Semiconductors.

@S-March Can you please give me some small hints here. What setup did you use in order to program you watch? I am so close right now. I've 3d printed the case, I´ve built the PCB, I've waited a million years for the battery and display to be delivered. Soo close....

S-March, 2019-08-18:
Hi there @jacobandersson86! Sorry it's taken me a while to get back. You will need to load the code over JTAG using the other set of 4 pads. You can hook up the JTAG lines to the SWD, CLK, RST and GND pads.

jacobandersson86, 2019-08-21:
No worries @S-March. I am so thankful that you are taking your time. I am using SmartSnippets Studio and I am getting my J-link to connect now. 

I do still have some problems though. I am still unable to program and I can't read. It could of course be a bad soldering or similar. 

I just have some questions to sort out some of my issues for the troubleshooting.

Q. How do you get the Vref for your JTAG device? I am using a Segger J-link and it needs V-ref to accept the connection. I'am also powering my watch from the batterypads at the moment since I don't have soldered the battery yet, hence I am using the VBUS also as my reference voltage for the J-link. 

Q. Are you using the SmartSnippetsStudio and the external tool called "program_qspi_jtag_win"? That is how I try to program the watch. I´ve also tried the tool from segger called J-Flash, but there are no presets there for the flash memory, so I didn't manage to flash with that tool either. 

Q. What bootloader file are you using? When I start the programmer in SmartSnippets Studio mentioned above, it says "bootlader file not specified, using internal uartboot.bin"

Many thanks in advance

rickmaity, 2019-09-02:
@jacobandersson86 have u found any solutions to your problem ....?

jacobandersson86, 2019-09-04:
Hi @rickmaity,
Unfortunatley I haven't. But I haven't had time to investigate this further. 

I think the key is to use the correct Vref for you J-Tag device. SWD needs SWCLK, SWIO, GND and Vref. In this project, the Reset has been populated as a pad, probably indicating that the reset needs to be controlled as well. I've tried finding a hardware reference guide for the SWD interface without success for this microcontroller. I might have some more time for this later this week. I'll get back if there is any success.

dannygrob, 2019-09-13:
@S-March +1 I'm also ready, pcb populated and tested, but somehow It won't recognize the ic. Connected the SWD to this pinout IO/CLK/RST/GND of the mini:

https://www.segger.com/fileadmin/images/products/J-Link/Accesory/Adapters/J-Link_9-pin_Cortex-M_Adapter.png

dannygrob, 2019-09-15:
Ah I got it to work. I connected VREF to VBAT (r6 i believe) and the 5v from the jtag to the vbus (not sure if this is needed). I got it to compile, upload and run the code, and the device shows up on my iphone as 'smarchwatch'. I can't get anything to display, but I did not upload any images yet (not yet sure how, and missing the font image)

Dirty-table, 2020-01-09:
Hi, I've been trying to build this smart watches for quite some time, but now I'm stuck at following problem and I need help with it: 

Once I connect programmer to the PC, it gets correctly recognized. But when I try to connect to it via J-link GDB server I get an error message: "WARNING: RESET (pin 15) high, but should be low. Please check target hardware." The program is constantly pulling RESET pin to high, once I launch it, no matter what I do. I've tried disconnecting RESET pin, which resulted in different error message saying that it cannot open GDB interface, and start GDB server.

Thank's for your help.

dannygrob, 2020-01-09:
Hi,

Using smart snippets and the build in debugging I did not use GDB server at all I believe. I load the software using the 'scripts' you have to include (you need to follow the tutorials and getting started, see https://github.com/S-March/smarchWatch_PUBLIC/issues/37, esp the getting started https://www.dialog-semiconductor.com/sites/default/files/um-b-047-da1468x_getting_started_7v0.pdf and section 8.3 Software Build)

I use the program_qspi_jtag to load the firmware, then use Ozone (included in the smartsnippets environment) to restart and debug. 

Hope that works.

BTW I rewrote the whole smartwatch code and made a new case for it..., if anyone is interested :)

Dirty-table, 2020-01-12:
Hi,

I've tried loading firmware through program_qspi_jtag_win, but every time I do it, I end up with error message: cannot open gdb interface. Any tips on what should I try next?

Thank's for the answer.