S-March/smarchWatch_Public #26: Panelized stencil file.

vanbluca, 2019-05-04:
@S-March could you put a few copies of the stencil files in a single document?  
Let’s say the stencil can be 400*240mm big. Could you put like 8 copies on the 400mm side and 5 copies on the 240mm side (so there are a total of 40 copies in a single 400*240mm stencil). This will cut the costs a lot!

S-March, 2019-05-10:
I will see what I can do. I am not sure I can do that within the next few days, but I will try.

vanbluca, 2019-05-11:
@S-March it would be highly appreciated! This way I could order 40 copies for a total $40 instead of $6 each! These stencils will be listed on eBay along with the PCBs (which have been shipped out today!).

ppyuri, 2019-06-04:
Hi @S-March, I really love your work. It is so thoughtfully designed without excess and beautiful. I want to build my own Smarch Watch! I'm just concerned about the soldering portion as I'm just beginning to get into electronics. Sorry for the beginner questions. How do I get a stencil for the watch and charger?

@vanbluca I saw your ebay posting for the boards and will probably purchase from you. Are these stencils available yet?

Semperubi77, 2019-10-15:
anyone know if the stencils are available?  
Thanks