S-March/smarchWatch_Public #28: Only for iPhones?

Greger312, 2019-05-07:
So I read in one of the issues that this watch is only for iPhones? Is there any more information about that, is it completely impossible to use it with android?

S-March, 2019-05-07:
It is not physically impossible. I just designed the watch for me and my personal use, so I made it for iOS. 

The firmware is completely open source though, so if you or someone else wants to add Android support I would gladly merge it into the project!

Greger312, 2019-05-07:
Do you know where i can find some information about how android support works? How hard would you estimate it to be? Possible for a mediocre programmer with zero background in android wearables?

S-March, 2019-05-07:
I don't have any resources on hand, but I would start googling for terms like "ANCS + Android" and see what comes up.

closed 2019-05-10 by S-March