S-March/smarchWatch_Public #30: Include updated BOM with pricing.

babarrett, 2019-05-15:
I looked up the Digikey prices for the BOM parts. I know these can change over time, but I added them to the BOM.xlsx file. 
Worthwhile, I think, to give people an idea of the cost to build.

(Enclosed)

[BOM.xlsx](https://github.com/S-March/smarchWatch_PUBLIC/files/3184606/BOM.xlsx)

vanbluca, 2019-05-16:
Have you also added the components for the charger in it (they are not listed in the current version). 
I described in issue #23 what components are needed (I believe it were only 3 components). In an updated version of the BOM.xlsx file I would really see those items added to it to avoid people buy not enough components.

babarrett, 2019-05-16:
Got it. Here's the new file.
[BOM.xlsx](https://github.com/S-March/smarchWatch_PUBLIC/files/3185580/BOM.xlsx)

vanbluca, 2019-05-16:
@babarrett thanks a lot! Would be great if @S-March updated the old one with this new version.
Oh and, not entirely necessary though, but you could also add the link to the eBay listing of the PCBs (https://ebay.us/FQBGIs) and the eBay link of the LCD screens (https://ebay.us/zBfP2e) under the “DigiKey number”. This could clarify the source. 

Side note for ordering the LCD screen:
I recommend to contact the seller and ask to ship it in a letter. This cuts both the very high shipping costs ($41) and import charges. Also their shipper (DHL) asks $20 administration costs for the import charges. This means you would have to pay $10+$41+$20+import charges for only 1 LCD. When they ship it in a letter, shipping will definitely be way less than $41. Also most companies put a value like $5 on an envelope so you won’t have to pay any import charges and administration costs!