S-March/smarchWatch_Public #34: Something interesting in Schematic file

ValentiWorkLearning, 2019-06-02:
In  Schematic file I noticed, that display's WR pin has't any connections on board and display driver uses the 9-bit SPI for transfering data between display  and MCU.
In ST7789V datasheet I've found that WR pin used for selecting, what would be transfered - command or data.
So, Why do you use the 9-bit SPI for display communication and where you find information about the display's interface?
[image]
I've created this issue because have a similar display, but haven't got any succesful results because trying to use 8 bit SPI with selecting data and comannd with toggling the WR pin.
Thanks for answer!

S-March, 2019-06-03:
Hi there!

I actually ended up using 9-bit SPI because I spent two weeks trying to get the display to work with the Data/Command line toggle and it wasn't working. I was banging my head against the wall and after pouring over the datasheet for the driver, noticed it had the option to do 9-bit SPI. The display would pick either 3 wire SPI, 4 wire SPI, a parallel approach or use of the Data/Command lines depending on how some internal (to the display). I had been looking at the data packets on my logic analyzer and couldn't find anything wrong with what I was doing.

Since I had nothing to lose, I tried the 9-bit approach and voila! The initialization code I had ran and my display turned on. 

The only way I figured it out was by dumb luck. There was no documentation for it anywhere. 

Page 33 of [this datasheet](https://www.newhavendisplay.com/appnotes/datasheets/LCDs/ST7789V.pdf) talks about the selection of the communication mode and then page 63 talks about how to pack the data.

Hope that helps!

ValentiWorkLearning, 2019-06-03:
Thanks a lot for your answer!
Therefore, interface with 3 and 4 wires is true random that defined only by display vendor.
I'll try one more with 9-bit SPI. Think, it must be successful. Thank you!
It will be nice, if I could contact with you on Facebook/etc, if possible.

ValentiWorkLearning, 2019-06-05:
Please, may you attach the waveform from logic analyser for unit sequence?