S-March/smarchWatch_Public #35: Antenna

jacobandersson86, 2019-06-13:
I assembled and soldered the PCB tonight. I noticed a few discrepancies in the material for example that the connector for the screen was missing in the BOM and that some other components were not populated. I perfectly understood why there weren’t any component values for the PI matching net for the antenna, but is that little blub the antenna itself or are you supposed to solder on a small wire or similar to use as antenna?? 

Q. Are you supposed to add an antenna? 

Another thing I can’t get my head around. The glass you are using, did it come with the screen? My screen hasn’t arrived yet and unfortunately I ordered the one with touch. Will give it a try and see
If it can be used whiteout having the cable for the touch conntcted.

Q. Do I need to buy/make a glass? 

Many thanks for the time you spent on this awesome project. If I manage to program my pcb in the coming week/weeks then I will contribute with the programming tutorial many has asked for. 

All the best!

abisheksethu, 2019-06-14:
[image]
**DF37B-24DP-0.4V(51)** - connector for screen without touch. isn't the right connector ?

S-March, 2019-06-14:
Hi @jacobandersson86 

With regards to the antenna, you will need to solder a piece of wire to the half hole - 31.25mm in length.

You are absolutely right, I forgot to include the watch glass. It can be found here: [35mm diameter 0.8mm thick](https://www.esslinger.com/watch-crystal-flat-round-mineral-glass-crystal-80-mm-thick/). It should cost $1.25 I believe.

As for the connector, @abisheksethu is correct. It is the [mating part](https://www.digikey.com/product-detail/en/hirose-electric-co-ltd/DF37NB-24DS-0.4V-51/H11850CT-ND/2230650) for the one on the display.

jacobandersson86, 2019-06-17:
Nice, thanks for the help and the quick replies!

closed 2019-06-17 by jacobandersson86