S-March/smarchWatch_Public #40: I've created an alternative version based on SmarchWatch

dannygrob, 2020-01-09:
Hi

I've created a new version of the watch, but with touchscreen. I've also changed the housing to be printed on my resin printer for a nice look. I noticed that this project is missing a lot to actually make it work (it has the basic BLE and some display routines, but even backlight code is missing). 

My version implements the display code a bit faster, adds touch support and has the code for the LIS3DH i2c chip. Also created time display and a menu. I struggled a lot, maybe we can help each other a bit more since a lot of people are interested in this watch. It is far from finished too, I mainly have a lot of issues debugging it. The tasks seem to stop, and I can't get the accelerometer and touch working at the same time. Also a real RTC is missing, and the sleep is broken atm. But take a look!

I'm working on changing the altium files to add the touch connector and fix some issues this one has (for ex. missing vref for debug and the vibrate motor 330ohm is not needed or it won't work)

https://github.com/dannygrob/OpenWatch

Danny

vanbluca, 2020-01-09:
This is awesome! Can’t wait to see more results!

d3mon187, 2020-01-24:
Hey @dannygrob , great to see someone else working on this project.  I'm a programmer that likes to get into things over his head, and this seems perfect.  I've got all the materials for the original project, so right now I'm working on figuring out how to solder everything and currently trying to figure out where everything goes.  Saw that you at least attempted to convert the pcb files to kicad, but couldn't figure out if you succeeded.  I think I went through the same steps as you, but once the project was in kicad it just threw errors on everything.  Did you manage to get it working, or did you switch to Altium?  I'm hoping to get away with something free, at least until I get my feet wet.  Altium lists upconvert as a website for hobbyists, but it doesn't seem to work for importing Altium either.

dannygrob, 2020-02-01:
I have a friend who had Altium, he just opened it there and modified the pcb. I think the original was Kicad though