S-March/smarchWatch_Public #41: Wrong Ohm listed on Schematic or BOM

d3mon187, 2020-01-24:
Loaded up the atrium files in attrium reader, and noticed that it states R12 as being 0.39ohm.  The BOM sheet also says 0.39ohm but the digikey part number is for a 390ohm resister.  Is the part number incorrect, or is the schematic wrong?  Is there one of the other resistors that will work in it's place?  I've already laid everything out on the pcb, so it would be quite a pain to wait at this point.

Also, you people are insane neurosurgeons!  It will be an absolute miracle if I manage to get these specs of dust soldered to the board successfully.  LOL

Thanks!

S-March, 2020-01-25:
It should be 0.39 Ohm, it looks like that was an error in the output to the digikey part number. Unfortunately it needs to be that small one to do the battery life measurements. Good luck!

closed 2020-01-25 by S-March

d3mon187, 2020-01-25:
Thanks, would this be the correct part to update the BOM list?  - https://www.digikey.com/product-detail/en/yageo/RL0805FR-070R39L/311-.39SCT-ND/1037225  

I think I'll try and solder everything in place and just add this one when it comes in.  Otherwise I'm sure I'll loose a part.  Will probably order extras too just in case.  I really didn't grasp how small some of this stuff is from the pictures and already lost a capacitor that I dropped and then completely vanished.